WLAN-Pi/v2-migration-docs #13: AAP22 - isc-dhcp - configure for WLAN Pi env

wifinigel, 2020-04-25:
1. Requires initial config of dhcpd.confg
2. Required for hostspot & wconsole
3. **Query** : Will USB0 now use APIPA IP address or DHCP assigned?"

wifinigel, 2020-05-05:
Requires details of dhcp scopes etc. to be migrated from v1.x image - assigned to Jerry

danielmundi, 2020-05-13:
Current configuration files are setup for RNDIS (ethernet over OTG).

```
$ cat /etc/default/isc-dhcp-server 
DHCPDv4_CONF=/etc/dhcp/dhcpd.conf
DHCPDv4_PID=/var/run/dhcpd.pid
INTERFACESv4="usb0"

$ cat /etc/dhcp/dhcpd.conf
# usb0 DHCP scope
subnet 169.254.42.0 netmask 255.255.255.224 {
  interface usb0;
  range 169.254.42.2 169.254.42.30;
  option domain-name-servers wlanpi.local;
  option domain-name "wlanpi.local";
  option routers 169.254.42.1;
  option broadcast-address 169.254.42.31;
  default-lease-time 600;
  max-lease-time 7200;
}
```

Configured on image customization: https://github.com/danielmundi/build/blob/b1ace37228e8e4906c9ac7c020effb274d1f9c1a/userpatches/customize-image.sh#L73-L93

wifinigel, 2020-05-13:
@jolla Jerry, do you have any input on this one? I lost track of what we decided on this.

jolla, 2020-05-15:
That looks correct. Unless someone says otherwise, this would be the proper way to do a link-local address

wifinigel, 2020-05-16:
@jolla Was the link-local just for usb0? The DHCP ranges for hotspot mode and wireless console will remain as-is for the new image, or do we need to re-visit those too for a standardised approach with a new 192.168.x.x range? They currently use 192.168.88.x/24 and 192.168.42.x/24 respectively...does it even matter as they are assigned when the mode is switched?

wifinigel, 2020-06-12:
@jolla @danielmundi I think we can close this one? I think the decision has been made to use the 169.254.42.1 address as standard?

wifinigel, 2020-07-15:
Complete as wconsole/hotspot/OTG working fine.

closed 2020-07-15 by wifinigel