WLAN-Pi/v2-migration-docs #3: BIB30 - Create initial base image build (inc msic-files repo configs)

wifinigel, 2020-04-25:


danielmundi, 2020-05-12:
Image has OTG and basic FPMS working - as well as several default packages like iperf, isc, hostpad, iw, etc.
Most scripts on FPMS still don't work because of absolute paths pointing to the old installation directory "/home/wlanpi". We are in the process of updating that.

wifinigel, 2020-05-13:
fpms has been updated to v2.0.0 to use relative paths for better portability. Some script paths called to activate external utilities (e.g. wconsole, hostspot, wiperf etc.) are still hard-coded at this point, as their installation locations are still fixed.

wifinigel, 2020-07-31:
The outstanding FPMS testing will cover off the final point raised in this issue. Closing as no longer required.

closed 2020-07-31 by wifinigel