WLAN-Pi/v2-migration-docs #5: BIB50 - Ensure team members have build capability or base image access

wifinigel, 2020-04-25:


danielmundi, 2020-06-09:
Current process for creating an Armbian build for WLAN Pi on a Ubuntu 18.04:

```
$ git clone -b master https://github.com/danielmundi/build.git
$ cd build
$ ./compile.sh
```

If you don't have a native Ubuntu 18.04 PC, you can use a virtual machine. Armbian already provides an easy way to integrate the build with Vagrant (configured to use Virtualbox as a virtualization provider).

In order to build with Vagrant, there are some extra steps to get it installed and configured:

```
### Install Vagrant from https://www.vagrantup.com/downloads.html

### Install plugin to allow resizing Vagrant disk
$ vagrant plugin install vagrant-disksize

### Download Ubuntu 18.04 VM
$ vagrant box add ubuntu/bionic64

### Once in a while run this to update the box (be aware that it might also break something hahaha)
$ vagrant box update
```

After that, you need to use below commands to start VM, log in to it and build:

```
$ cd userpatches
$ vagrant up
$ vagrant ssh

### now you are inside the VM
$ cd armbian
$ ./compile.sh
```

After finished, stop VM with `vagrant halt`.
To remove the image and free up the disk use `vagrant destroy`.

danielmundi, 2020-06-25:
For more up to date (and sometimes broken) build we can use **dev** branch.

```
$ git clone -b dev https://github.com/danielmundi/build.git
$ cd build
$ ./compile.sh
```

wifinigel, 2020-07-31:
Build access provided & used by team regularly - closed

closed 2020-07-31 by wifinigel