WLAN-Pi/v2-migration-docs #20: AAP60 - python3 - Verify present in image & update with apt update

wifinigel, 2020-04-26:


danielmundi, 2020-05-13:
Current packages being installed with apt:

```
libpython3-dev
libpython3.7
libpython3.7-dev
python3-certifi
python3-chardet
python3-dev
python3-idna
python3-numpy
python3-protobuf
python3-requests
python3-serial
python3-six
python3-urllib3
python3-usb
python3.7-dev
python3-scapy
python3-wheel
python3-pillow
python3-pil
python3-pip
python3-setuptools
python3-smbus
```

Might not need all of those and some are probably could only be installed on python virtual environments for specific packages.

wifinigel, 2020-05-13:
Other modules:
```
- https://pypi.org/project/speedtest-cli/ (used by wiperf)
- (believe this also installs /usr/local/bin/speedtest which is used by speedtest in fpms)
- manuf (used by profiler2)
- dataclasses (profiler2)
```

danielmundi, 2020-05-31:
@wifinigel, Jerry asked to include **speedtest** from Ookla, which is the official distribution.
The problem is that it says it doesn't play well being installed in the same system as **speedtest-cli**.

Would the official speedtest work with wiperf?

danielmundi, 2020-05-31:
The modules needed by **profiler** are installed as dependencies, so no need to add them here.

wifinigel, 2020-05-31:
@danielmundi I'd have to re-code wiperf I suspect. If the Ookla is running as a server and I'm running speedtest-cli as a client in Python, I'm not sure they will clash? Need to understand the issue in more detail.

danielmundi, 2020-05-31:
Actually speedtest by Ookla is also a client.
On the WLAN Pi image I posted yesterday you can just call `speedtest` to check how it works.

We can try to have both anyway and see what happens.

jolla, 2020-06-01:
Apparently I incorrectly assumed we were using the official Speedtest CLI package, if there is a benefit to using the unofficial speedtest-cli package we can switch to that

wifinigel, 2020-06-01:
Guys, which package is the "official" Speedtest...I'm not clear. Is it this package that was mentioned in Slack? https://github.com/miquels/speedtest

We have 3 areas where a speedtest is performed:

1. On the current web GUI that operates as a HTML5-based speedtest server (this uses this project: https://github.com/librespeed/speedtest/)
2. In wiperf, we perform a speedtest using the speedtest-cli client as a Python module (https://pypi.org/project/speedtest-cli/)
3. In FPMS, we also uses the same speedtest-cli client as a CLI command (as far as I know)

The new utility proposed by Jerry would just be for case (1)?

jolla, 2020-06-01:
I was referring to the official Ookla Speedtest.net CLI client - https://www.speedtest.net/apps/cli

The unofficial speedtest-cli existed before the official Ookla Speedtest CLI was released - https://github.com/sivel/speedtest-cli

I had assumed you were using the official Ookla Speedtest CLI but apparently I was wrong and you are still using the unofficial speedtest-cli.

Also, I don't see any reason both can't be installed, other than extra bloat.

wifinigel, 2020-06-01:
Sorry Jerry, I didn't know about the Ookla official package. Would there be any legal considerations if we included it in our distribution?

jolla, 2020-06-01:
Good point, for now lets stick with using speedtest-cli for the purpose you have been using it. @danielmundi please leave out the official Ookla Speedtest CLI for now and install speedtest-cli.

danielmundi, 2020-06-03:
This is changed, just testing before pushing to repo.

danielmundi, 2020-06-03:
This isn't the best place to discuss this, but since we already on the topic.

@wifinigel,
Do we need to keep https://github.com/librespeed/speedtest/ for v2.0?
I saw it was included as part of #46, but wlanpi-topping it was cancelled.
Although HTML5 speedtest it's still marked as Retain in our spreadsheet.

wifinigel, 2020-07-31:
OK, the librespeed code is used for HTML 5 speed testing in web GUI. This is working, so can be left in as-is. @danielmundi

closed 2020-07-31 by wifinigel