WLAN-Pi/v2-migration-docs #24: AAP76 - iperf3 - Run test suite to verify operational

wifinigel, 2020-04-26:


wifinigel, 2020-07-02:
Testing performed:

Rig:

1. Silver WLANpi
2. Image: Armbian_20.08.0-trunk_Wlanpi-neo2_buster_current_5.4.49.img.gz
3. Ether connection to 1Gbps Cisco c3650 switch port
4. Windows 10 laptop (wireless disabled) - connected to 1Gpbs Cisco switch port

Tests run:

1. Test from WLAN Pi fresh boot - no config mods
2. TCP iperf3 test:

[image]

3. UDP iperf3 test:

[image]

Both results consistent with expected performance. No modifications required to firewall.

4. Stopped iperf service with cockput GUI. Appeared to create failure msg:

[image]

@danielmundi Is this an issue that needs fixing in the start/stop script for iperf3?

wifinigel, 2020-08-25:
@danielmundi Had chance to look at this?

jolla, 2020-08-25:
Duplicated the "failed" error in both cockpit and from CLI, however also confirmed it is indeed working as expected (minus the error message), the service is starting and stopping iperf3 successfully. Closing for now, we can investigate the error message later.

closed 2020-08-25 by jolla