WLAN-Pi/v2-migration-docs #29: TPP00 - wifiexplorer-sensor - Add to base image build process

wifinigel, 2020-04-26:


wifinigel, 2020-07-05:
Instructions for this can be found in this repo from @adriangranados : https://github.com/adriangranados/wifiexplorer-sensor

@danielmundi may be worth checking with @adriangranados that this is the current info

adriangranados, 2020-07-06:
That's the current info. The sensor script should work with Python 3. Scapy is the only dependency. In the repo you can find the System V startup script. We would need to write a systemctl startup script.

adriangranados, 2020-07-15:
I created the systemd startup script (see scripts/wifiexplorer-sensor.service in sensor's repo). I also tested the latest version of the sensor script and it's working fine.

@danielmundi What else is needed to "debianize" and install the sensor in the base image?

adriangranados, 2020-07-16:
By the way, the new repo URL is [https://github.com/intuitibits/wifiexplorer-sensor](https://github.com/intuitibits/wifiexplorer-sensor)

adriangranados, 2020-07-21:
@danielmundi has taken care of this. I tested it and the service loads fine, sensor is available for scanning provided an adapter is connected.

closed 2020-07-21 by adriangranados