WLAN-Pi/v2-migration-docs #40: PHP00 - fpms - Add to base image build process (inc debian pkg creation & venv)

wifinigel, 2020-04-26:
1. During image build, package needs to be pulled in from GitHub repo, Debianized and then installed. 
2. Systemctl scripts are required to allow on-demand start/stop of application from FPMS/web gui/Cockpit
3. Existing repo still needs review due to previous ommissions found by Jerry previously

danielmundi, 2020-05-13:
FPMS update:

- Starting and showing menu options.
- Daemon is ported to systemd service instead of rc.local.
- Added Makefile to easy the installation process - `make && make install`

Since the package is installed on a new location, scripts still need to change hard coded paths to work properly. This is in progress.

wifinigel, 2020-05-13:
Initial release of proposed code for fpms available at https://github.com/WLAN-Pi/fpms/releases/tag/v2.0.0. Updates/fixes to this will be applied to master branch

Is the fpms python script running in its own venv as part of the build process?

danielmundi, 2020-05-13:
It's not running on a venv yet.

The tricky is that I believe any scripts executed from FPMS will also have to be on the same venv.
Also, we have some forking on the process, so need to check how venv handles that.

We could work on that after we get the base image ready.

danielmundi, 2020-05-17:
@wifinigel can we close this task and open another one to make the venv?

wifinigel, 2020-05-18:
Created new task & closing this one:
https://github.com/WLAN-Pi/v2-migration-docs/issues/69

closed 2020-05-18 by wifinigel