WLAN-Pi/v2-migration-docs #41: PHP03 - fpms - Create systemctl scripts for start on boot & manual startup/shutdown of server process

wifinigel, 2020-04-26:
1. During image build, package needs to be pulled in from GitHub repo, Debianized and then installed. 
2. Systemctl scripts are required to allow on-demand start/stop of application from FPMS/web gui/Cockpit
3. Existing repo still needs review due to previous ommissions found by Jerry previously"

danielmundi, 2020-05-13:
FPMS service is enabled.

So far it's running as a forked process, because the original daemon forked itself. In order to have systemd mange this daemon, we need to configure it with the PID file, so it knows under which PID the process is running.

The ideal is to have systemd take care of the forking process. This is will became a new task later on.

danielmundi, 2020-05-27:
Merged modifications to main fpms repo.

closed 2020-05-27 by danielmundi