WLAN-Pi/v2-migration-docs #42: PHP06 - fpms - Run test suite to verify operational

wifinigel, 2020-04-26:


jiribrejcha, 2020-08-19:
All menu items tested successfully with these caveats:
- Public IP address: output does not scroll down, refreshes too often and sometimes fails displaying the output, one for me to pick up
- Wiperf mode: switching not working "Switch failed: wiperf", @wifinigel is this expected (I have not applied any wiperf config or other changes)?
- Profiler: when you start it it gets stuck at "Info: Please wait..." and does not enable Profiler, no way to exit this screen, no SSID is broadcast, @joshschmelzle could you look into this please?

jiribrejcha, 2020-08-19:
**Fixed and did PR to FPMS dev**
Public IP address: output does not scroll down, refreshes too often and sometimes fails displaying the output, one for me to pick up

joshschmelzle, 2020-08-19:
I will take a look later tonight after my day job.

jiribrejcha, 2020-08-19:
@joshschmelzle there is no /etc/profiler2 folder on my test Pi.

My image was built before the merge of this code to builder:
copy_overlay /etc/profiler2/config.ini -o root -g root -m 644

Building a fresh one and will re-test.

joshschmelzle, 2020-08-19:
@jiribrejcha i am guessing `copy_overlay /etc/profiler2/config.ini -o root -g root -m 644` must be failing if the destination does not exist. i think we may need to add `mkdir /etc/profiler2` if `copy_overlay` does not handle creating the folder for us.

jiribrejcha, 2020-08-19:
@joshschmelzle Profiler is now working, no changes needed. Sorry for bothering you. The latest PR fixed it and the config file is there in /etc/profiler2/config.ini:

[image]

closed 2020-08-19 by jiribrejcha

joshschmelzle, 2020-08-19:
Awesome!