WLAN-Pi/v2-migration-docs #48: PHP30 - wlanpihotspot - Add to base image build process

wifinigel, 2020-05-01:


danielmundi, 2020-05-27:
The problem with current switch method is that it doesn't play well with systemd services to be managed by Cockpit. This is valid for all modes that just update the files and restart the PI.

The main problem is that we need to reboot the Pi in order for the service to be actually up and I'm not sure how to handle that.

I need to check a little further if it's possible to manage this with systemd or if would be better to change how we switch modes.

wifinigel, 2020-05-31:
Yes, this is a tricky one. The only way I can think of approaching this is to include a network restart after performing the file re-configuration needed for each mode switch. 

This would require the start/stop script to run our current mode switch script to do the relevant file copying/linking and then call a network services restart rather than a reboot. This is only an initial guess...I'm not sure if it would technically work. We'll have to do some testing.

wifinigel, 2020-05-31:
I'll take a look at hotspot and see if I can get it doing on the v1.9 image with no reboot initially. It could be we could just restart a few processes (e.g. hostapd, dhcpd) to make to them re-read their config files to make it work.

wifinigel, 2020-06-01:
@danielmundi I've spent quite a few hours trying to get hotpot to switch modes with a reboot. The issue is that there are quite a few services that are re-configured as well as a switch in forwarding mode between interfaces. I can _almost_ get it to work, but the number of services I have to restart is high and I'm not confident that networking is going to work correctly if people switch between modes. A full reboot is the only clean way I can get it to work and not have any concerns about the wider operation of the unit if a user switches between modes.

If possible, I'd like to maintain the current system of rebooting to switch mode to preserve to reliability of networking when flipping between modes, as all make fundamental changes to the network operation of the WLAN Pi. This obviously removes the option of  using a service stop/start from cockpit. I think a web-gui based, or FPMS switch is the best option.

Here is my switcher script to try to do things without a reboot...you can see the hoops I'm having to jump through in terms of service restarts (and this still doesn't fully work) : https://github.com/WLAN-Pi/wlanpihotspot/blob/v2_image/hotspot_switcher

danielmundi, 2020-06-03:
Hey @wifinigel, that's ok by me. I'll try to play around with this later on, but for now we can stick to what we know works.

One thing we could add to the scripts is to use some kind of lock file, that would register the current WLAN Pi state. This would at least allow us to probe for the status to display on web-gui and also try to prevent actions that wouldn't work on current mode.

I think a file `/etc/wlanpi-state` would be a good place to put it.

wifinigel, 2020-06-03:
Agreed. I'll create some new actions for the impacted modules (fpms, wconsole, hotspot & wiperf).

wifinigel, 2020-06-12:
@danielmundi hotspot code now updated for v2 image. Code is pushed to v2)image branch of repo: https://github.com/WLAN-Pi/wlanpihotspot/tree/v2_image.

Install process:
1. Copy repo files to /etc/wlanpihotspot
2. Move /etc/wlanpihotspot/hotspot_switcher to /usr/bin/hotspot_switcher & chmod 775

danielmundi, 2020-06-24:
This is included in the image, need to confirm if there is any other configuration to be made.

jiribrejcha, 2020-07-14:
Added by Daniel and working fine.

closed 2020-07-14 by jiribrejcha