WLAN-Pi/v2-migration-docs #51: PHP45 - wconsole - Run test suite to verify operational (inc mode switch)

wifinigel, 2020-05-01:


wifinigel, 2020-07-02:
Tests completed with Armbian_20.08.0-trunk_Wlanpi-neo2_buster_current_5.4.49.img.gz:

Test rig:
- Silver Neo2
- CF-912 NIC
- USB/Serial cable 
- Cisco 3560-CX switch serial console

1.  Switched from classic mode to wconsole using fpms - reboot took 43 secs 
2. fpms display shows wiperf mode OK and IP address of 192.168.42.1 on wlan0
3. SSID "wifi_console" detected OK from Windows laptop
4. Wi-Fi scanner shows SSID is AES/PSK on channel 1
5. Joined SSID with Windows laptop.
6. Performed telnet session to 192.168.42.1: 9600 and obtained serial port access to switch
7. Able to peform CLI commands as expected
[image]

8. Selected classic mode from FPMS. Time to boot back to classic mode: 47secs

wifinigel, 2020-07-15:
interfaces file update & re-test req.

jiribrejcha, 2020-07-15:
File updated and re-tested. Working fine after the change. Did PR to dev.

jiribrejcha, 2020-07-16:
Built new dev image and successfully re-tested.

closed 2020-07-16 by jiribrejcha