WLAN-Pi/v2-migration-docs #59: TPA30 - WIFiExplorer - Test WFE on Mac can still access WLAN Pi as data source

wifinigel, 2020-05-01:


wifinigel, 2020-07-15:
Asked Helge if he can look at this.

adriangranados, 2020-07-15:
I can test it.

adriangranados, 2020-07-15:
The latest version of the sensor script works fine. However, the ability to auto-detect the WLAN Pi as an "external adapter" using the OTG connection doesn't work anymore because the current version of WiFi Explorer Pro is looking for 192.168.42.1. I'll change WiFi Explorer Pro to look for 169.254.42.1 as well. And the other thing is the startup script. I'll work on that.

jolla, 2020-07-15:
Sounds good, we are still testing the waters using a link-local address, but it would be a good idea to add both.

Thank you,
Jerry Olla

adriangranados, 2020-07-21:
Tested it on latest image. There was an issue with Scapy (on the script side) that's been corrected. Will test again when the fix is made available in the next build.

adriangranados, 2020-07-29:
Verified as working in the latest image. Closing.

closed 2020-07-29 by adriangranados