WLAN-Pi/v2-migration-docs #64: MPR002 - Misc Project Req - root account password

wifinigel, 2020-05-05:
5th May 2020 - decided to have standard root password for the image, but disable root account for login. Choose a new root password for v2.x image (i.e. not same as 1.x image)

danielmundi, 2020-05-12:
Password set and root account disabled by setting shell to **/sbin/nologin**.

Another option we have is to just disable the password for root, this way the account will also be disabled and if someone does want to enable it, they have to set their own custom password.

danielmundi, 2020-05-13:
From [https://help.ubuntu.com/community/RootSudo#root_account](Ubuntu forums).
Password is disabled with: `sudo passwd -dl root`
Enable root setting the password with: `sudo passwd root`

wifinigel, 2020-05-13:
Do we need to have a conversation about this one to choose an option?