WLAN-Pi/v2-migration-docs #65: MPT01 - Check licenses of existing packages to ensure we are in alignment with permitted use

wifinigel, 2020-05-12:
Each package we use withtin the WLAN Pi project will have its own license that dicates how it may be used/distributed. We need to check each package and ensure we are not vilotating that license by any modifications we apply or how we re-distribute those packages.

danielmundi, 2020-05-13:
Some projects we use and their licences.

- [Armbian](https://github.com/armbian/build/blob/master/LICENSE): GPLv2
- [NanoHatOLED](https://github.com/friendlyarm/NanoHatOLED/blob/master/README.md): MIT
- [BakeBit](https://github.com/pragtich/BakeBit/blob/master/README.md): MIT

joshschmelzle, 2020-05-13:
I added the GPLv3 license to [my profiler port](https://github.com/joshschmelzle/profiler2). I'm willing to change the license if required.