WLAN-Pi/v2-migration-docs #79: BIB33 - Disable NetworkManager service in base build

wifinigel, 2020-06-12:
NetworkManager has caused some operational issues for wlan0 during mode switching. Disabling the NM service mitigates the issue. Please disable the service in the base build.

danielmundi, 2020-06-24:
Services disabled:

```
systemctl disable NetworkManager.service
systemctl disable NetworkManager-wait-online.service
systemctl disable NetworkManager-dispatcher.service
systemctl disable network-manager.service
```

jiribrejcha, 2020-07-14:
NetworkManager re-enabled only for Ethernet interfaces and it now compatible with the v2 image including Hotspot and latest build of Wi-Fi Console.

jiribrejcha, 2020-07-21:
Change of direction - we decided to disable NetworkManager completely and not to use it anymore

@danielmundi, can you please re-apply these changes to the v2 image?

```
systemctl disable NetworkManager.service
systemctl disable NetworkManager-wait-online.service
systemctl disable NetworkManager-dispatcher.service
systemctl disable network-manager.service

```

jiribrejcha, 2020-07-21:
The goal is for eth0 DHCP client leases to be available in /var/lib/dhcp/dhclient.eth0.leases for FPMS and other scripts

jiribrejcha, 2020-07-29:
NM removed from the image completely by Daniel
eth0 section uncommented in Classic mode already
Uncommented eth0 section in Hotspot and wconsole modes and awaiting PR approval to dev

jiribrejcha, 2020-07-29:
NM completely removed from v2 image by Daniel

closed 2020-07-29 by jiribrejcha