WLAN-Pi/v2-migration-docs #85: Dynamically assigned IP address on eth0 is sticky and does not release/renew

jiribrejcha, 2020-07-10:
After eth0 gets an IP address from a DHCP server it will not release it until you reboot the Pi. No matter how many times you disconnect the Ethernet cable or connect the Pi to another network which is uses a completely different subnet.

On v1.x we fixed this by installing ifplugd. Shall we do this again?

Config file:
/etc/default/ifplugd
And its content:
INTERFACES=“eth0"
HOTPLUG_INTERFACES=“wlan0”
ARGS=“-q -f -u0 -d0 -w -I”
SUSPEND_ACTION=“stop”

jiribrejcha, 2020-07-10:
I've just tested this on v2 and it is working fine and fixes the issue.

Install:
sudo apt install ifplugd

Config file:
/etc/default/ifplugd

And its content:
INTERFACES=“eth0"
HOTPLUG_INTERFACES=“wlan0”
ARGS=“-q -f -u0 -d0 -w -I”
SUSPEND_ACTION=“stop”

jolla, 2020-07-10:
This appears to be due to disabling Network Manager. 
https://github.com/danielmundi/build/commit/937ed0eaaf4ed2aaae5cf79b3d32ebc1231975d3

We should look at re-enabling Network Manager, but restricting it to only managing wired interfaces

Perhaps using the [keyfile] method mentioned in this article work?
https://support.qacafe.com/knowledge-base/how-do-i-prevent-network-manager-from-controlling-an-interface/

Ex. 
[keyfile]
unmanaged-devices=interface-name:wlan*

jiribrejcha, 2020-07-13:
Enabling Network Manager fixed the IP address stickiness issue and I’ve successfully tested this configuration with Classic, Hotspot and Wi-Fi Console modes. Cockpit can see and manage the eth0 interface now. All other interfaces are configured to be unmanaged by Network Manager to prevent any wlan0 interface issues seen in the past.
The only thing I’ve noticed is DHCP running on the Pi in Wi-Fi Console mode takes 60-70 seconds to assign an IP address to a Wi-Fi client. In Hotspot mode the assignment only takes few seconds. Investigating now.

**Changes applied to a vanilla v2 dev image**

sudo systemctl enable NetworkManager.service
sudo systemctl enable NetworkManager-wait-online.service
sudo systemctl enable NetworkManager-dispatcher.service
sudo systemctl enable network-manager.service
sudo nano /etc/NetworkManager/NetworkManager.conf

Add a new section:
[keyfile]
unmanaged-devices=*,except:type:ethernet

Start services:
sudo systemctl start NetworkManager.service
sudo systemctl start NetworkManager-wait-online.service
sudo systemctl start NetworkManager-dispatcher.service
sudo systemctl start network-manager.service

jiribrejcha, 2020-07-13:
With Network Manager enabled, both Hotspot and Wi-Fi Console mode are now working just fine and I managed to add a couple new features to Wi-Fi Console

**Included in this pull request (wconsole dev branch)**
- wconsole and hotspot now work with v2 Network Manager, tested with latest dev v2 image, NM instructions below
- when eth0 is connected in Wi-Fi console mode, you can access the internet and online documentation while using using Wi-Fi or wired console
- added 115200 baud rate listening on ports 11520-11528 (note missing zero in port number)

**How to install latest wconsole (after my PR gets approved)**
1. Clone latest wconsole dev branch to /etc/
2. Move mode switcher to its new place:
sudo mv /etc/wconsole/wconsole_switcher /usr/bin/wconsole_switcher

**Enable Network Manager (NM)**
sudo systemctl enable NetworkManager.service
sudo systemctl enable NetworkManager-wait-online.service
sudo systemctl enable NetworkManager-dispatcher.service
sudo systemctl enable network-manager.service

**Edit config file**
sudo nano /etc/NetworkManager/NetworkManager.conf

Add a new section:
[keyfile]
unmanaged-devices=*,except:type:ethernet

**Start NM services**
sudo systemctl start NetworkManager.service
sudo systemctl start NetworkManager-wait-online.service
sudo systemctl start NetworkManager-dispatcher.service
sudo systemctl start network-manager.service (edited)

jiribrejcha, 2020-07-14:
Fixed my enabling Network Manager for Ethernet interfaces only. Details above.

Latest Wi-Fi Console is now compatible with NM and Hotspot already was. No other issues found after enabling NM.

closed 2020-07-14 by jiribrejcha