WLAN-Pi/v2-migration-docs #88: NetworkManager requirement - remove all references to eth0 from /etc/network/interfaces

jiribrejcha, 2020-07-15:
If we keep eth0 config in /etc/network/interfaces, there will be conflicts and eth0 will get 2 dynamic IP addresses assigned to eth0. Removing eth0 from /etc/network/interfaces fixes this. Tested.

jiribrejcha, 2020-07-15:
Suggested new content of /etc/network/interfaces:

```
#WLAN Pi now uses NetworkManager to manage eth0, eth0 section has to be commented out here to prevent 2 IPs from being assigned to eth0 and other conflicts
# Default eth0 as DHCP
#allow-hotplug eth0
#iface eth0 inet dhcp
```

wifinigel, 2020-07-15:
@jiribrejcha I'll need to re-test wiperf, as it uses an eth0 config in this file.

wifinigel, 2020-07-15:
@jiribrejcha wconsole & hotspot also will need cfg files modifying & retesting as  their interfaces file uses an eth0 entry

jiribrejcha, 2020-07-15:
@wifinigel I've tested wconsole and hotspot now. Both are working fine after commenting eth0 config out in /etc/network/interfaces.

jiribrejcha, 2020-07-15:
Tested and did pull requests to:
- wconsole dev
- wlanpihotspot v2_image

@danielmundi could you please update /etc/network/interfaces in the base image and comment out the eth0 section, please? This is to allow NetworkManager properly manage eth0 without conflicts.

Suggested eth0 section content in /etc/network/interfaces after this change:

#WLAN Pi now uses NetworkManager to manage eth0, eth0 section has to be commented out here to prevent 2 IPs from being assigned to eth0 and other conflicts
# Wired ethernet
#auto eth0
#iface eth0 inet dhcp

jiribrejcha, 2020-07-21:
Change of direction - we are disabling NetworkManager completely on the WLAN Pi v2 now

closed 2020-07-21 by jiribrejcha