WLAN-Pi/v2-migration-docs #89: Profiler crashes FPMS

jiribrejcha, 2020-07-17:
Menu > Profiler > Status or Start crashes FPMS on the latest dev v2 NEO2 image.

jiribrejcha, 2020-07-17:
[image]

joshschmelzle, 2020-07-17:
[FPMS #11](https://github.com/WLAN-Pi/fpms/pull/11) _should_ fix this.

jiribrejcha, 2020-07-21:
@joshschmelzle thanks, unfortunately that did not fix it.

joshschmelzle, 2020-07-21:
@jiribrejcha 

when you run `sudo journalctl -u fpms` do you have similar to the following output?

```
Jul 21 11:17:27 wlanpi oled-start[904]: Traceback (most recent call last):
Jul 21 11:17:27 wlanpi oled-start[904]:   File "bakebit_nanohat_oled.py", line 638, in <module>
Jul 21 11:17:27 wlanpi oled-start[904]:     time.sleep(1)
Jul 21 11:17:27 wlanpi oled-start[904]:   File "bakebit_nanohat_oled.py", line 537, in receive_signal
Jul 21 11:17:27 wlanpi oled-start[904]:     g_vars['key_mappings'][ g_vars['key_map'] ]['key_actions']['key2']()
Jul 21 11:17:27 wlanpi oled-start[904]:   File "bakebit_nanohat_oled.py", line 311, in menu_right
Jul 21 11:17:27 wlanpi oled-start[904]:     button_obj.menu_right(g_vars, menu)
Jul 21 11:17:27 wlanpi oled-start[904]:   File "/usr/share/fpms/BakeBit/Software/Python/modules/nav/buttons.py", line 63, in menu_right
Jul 21 11:17:27 wlanpi oled-start[904]:     g_vars['option_selected']()
Jul 21 11:17:27 wlanpi oled-start[904]:   File "bakebit_nanohat_oled.py", line 248, in profiler_status
Jul 21 11:17:27 wlanpi oled-start[904]:     app_obj.profiler_status(g_vars)
Jul 21 11:17:27 wlanpi oled-start[904]:   File "/usr/share/fpms/BakeBit/Software/Python/modules/apps.py", line 171, in profiler_status
Jul 21 11:17:27 wlanpi oled-start[904]:     self.profiler_ctl(g_vars, action="status")
Jul 21 11:17:27 wlanpi oled-start[904]:   File "/usr/share/fpms/BakeBit/Software/Python/modules/apps.py", line 75, in profiler_ctl
Jul 21 11:17:27 wlanpi oled-start[904]:     sys.path.append('/opt/wlanpi/pipx/venvs/profiler2/lib/python3.7/site-packages')
Jul 21 11:17:27 wlanpi oled-start[904]: NameError: name 'sys' is not defined
```

joshschmelzle, 2020-07-21:
i just submitted a PR to fix the this, i think. https://github.com/WLAN-Pi/fpms/pull/12

jiribrejcha, 2020-07-28:
Re-tested on the latest image. Working fine now.

closed 2020-07-28 by jiribrejcha